CrossTheRoadElec/Phoenix5-Documentation #3: Missing items from migration guide

sciencewhiz, 2018-01-08:
Here are the items I had to change to get our 2017 code to compile with Phoenix, that were not already included in the migration guide.

setPID - replaced by individual config methods
getSetpoint - can't find replacement
getClosedLoopError - added parameter
getOutputVoltage - getMotorOutputVoltage
ControlMode - PercentVBus renamed PercentOutput and Speed renamed velocity.

sciencewhiz, 2018-01-09:
Thanks for the update. Does this mean there isn't a replacement for getSetpoint?

AaronPinto, 2018-01-09:
Also what about a replacement for setAllowableClosedLoopErr()?

JCaporuscio, 2018-01-09:
@sciencewhiz Not currently.  A few people have asked for it back, so we're going to be adding it with this week's motion profile update.

@AaronPinto Apologies, it looks like it was missing from the migration guide.  It has been updated accordingly.

NoMod-Programming, 2018-01-10:
Also, what happened to `setSafetyEnabled(boolean)` and `setClosedLoopOutputDirection(boolean)`?

JCaporuscio, 2018-01-13:
@NoMod-Programming 
setSafetyEnabled is part of WPI_TalonSRX, as that is the class that implements WPI interfaces.
setClosedLoopOutputDirection is now in the migration guide.

@sciencewhiz We added getClosedLoopTarget in the latest release as a replacement for getSetpoint, and it is now documented in the migration guide.

closed 2018-01-13 by JCaporuscio