CrossTheRoadElec/Phoenix5-Documentation #6: CANTalon methods not covered in migration guide...

kensthilaire, 2018-01-08:
Can you provide guidance on some of the methods that were available on the CANTalon but are not available with the WPI_TalonSRX class? The migration guide seems to only cover a subset of the changes.

I was attempting to have a set of students migrate our 2017 robot code to the new 2018 tools and libraries but looking at the extent of the errors/changes, this looks as though it will be a waste of time and energy. Was there any consideration given to provide backwards compatibility to the CANTalon interface?

Thank you, and I appreciate any guidance you can provide.

Ken St.Hilaire
Mentor FRC 1073

JCaporuscio, 2018-01-09:
I'd really like to understand where you're running into trouble.

Are there specific sections/examples of code that your team has run into difficulty porting?  If you're not comfortable sharing your code in a public setting, feel free to e-mail our support (support@ctr-electronics) - we keep code sent to us confidential.

A backwards-compatible CANTalon interface is pretty difficult.  There are a couple core concepts (like the way we do motor inversion) that just don't translate well from last year to this year.  A partial porting is possible - there's a version of it [here](https://github.com/CrossTheRoadElec/Phoenix-CANTalon-Sandbox), but it hasn't been tested.

kensthilaire, 2018-01-09:
Thank you for the response, Jacob. Since I initially filed this issue, I have noticed that the migration document has been recently updated to include some ways to deal with some of the troublesome areas (get/set encoder position/velocity).

But, here are some snippets of code that have errors on every line due to undefined methods. The 'driveTrainrightMotor3' is declared as a WPI_TalonSRX, having formerly been declared as a CANTalon. Specific guidance here would be much appreciated.

        RobotMap.driveTrainleftMotor3.setVoltageRampRate(motorRampRate);
        RobotMap.driveTrainrightMotor3.setVoltageRampRate(motorRampRate);

    	//setting PID for the right side
    	RobotMap.driveTrainrightMotor3.enable();
	RobotMap.driveTrainrightMotor3.setEncPosition(0);
    	RobotMap.driveTrainrightMotor3.setFeedbackDevice(FeedbackDevice.QuadEncoder);
	RobotMap.driveTrainrightMotor3.configEncoderCodesPerRev(360);
    	RobotMap.driveTrainrightMotor3.setProfile(0);
    	RobotMap.driveTrainrightMotor3.setF(0.1);
    	RobotMap.driveTrainrightMotor3.setP(1); //ideal p value from testing 0.485
    	RobotMap.driveTrainrightMotor3.setI(0); 
    	RobotMap.driveTrainrightMotor3.setD(0);
    	RobotMap.driveTrainrightMotor3.setAllowableClosedLoopErr(20);
    	RobotMap.driveTrainrightMotor3.reverseOutput(false);
    	RobotMap.driveTrainrightMotor3.configMaxOutputVoltage(8);
    	RobotMap.driveTrainrightMotor3.changeControlMode(ControlMode.Position);

All in all, I would have expected a cleaner migration to the new API, or at least a backwards compatible option for teams to quickly and easily migrate an existing robot code base to the new tool versions. We really don't have time to spend refactoring existing code just to use the new tool versions.

JCaporuscio, 2018-01-09:
It looks like most of these were present already, but there were a couple missing from the migration guide that I went back and added.  I'm going to go through the entire migration guide later today and make sure all of last year's methods are there.

enable() doesn't exist anymore (and isn't needed), so you can safely just remove that one.

kensthilaire, 2018-01-10:
Thank you for the additional follow-up, Jacob. I will go have a look at the latest migration guide.
Ken

closed 2018-01-10 by kensthilaire