CrossTheRoadElec/Phoenix5-Documentation #7: Documentation of StatusFrame defaults

jhh, 2018-01-09:
We speed up status frame rates for our telemetry tools while tuning motors so we'd like to revert them back to default rates afterwards. Documentation of new frame rates and a mapping of the old `CANTalon.StatusFrameRate` enum values to new `StatusFrame` enum values in the migration guild would be helpful also.

JCaporuscio, 2018-01-09:
Resetting the configuration to factory default will also reset the status/control frame periods.

https://github.com/CrossTheRoadElec/Phoenix-Documentation#factory-default-the-configuration-parameters

The default frame rates for status frames do need to be documented, though.
I can't give an exact ETA but I'm going to leave this issue open until the docs are updated.

ozrien, 2018-01-10:
These will be added to Phoenix-doc but for now...
STATUS1 General 10ms
STATUS2 Feedback0 20ms (selected sensor for primary pid)
STATUS3 (Quadrature) 160ms
STATUS4 (Analog, Temp, Vbat) 160ms
STATUS8 (Pulse width) 160ms
STATUS13 (PIDF0- error,derr,iaccum) 160ms
STATUS14 (PIDF1- error,derr,iaccum) 160ms

AaronPinto, 2018-01-15:
If I'm getting ERROR -3 CTR: CAN frame not received/too-stale. Talon SRX 15 GetSelectedSensorVelocity with a Talon SRX in CAN with a CTRE Mag Encoder plugged into it, what status frame rate should I change if at all any?

ozrien, 2018-01-15:
GetSelectedSensorVelocity is in STATUS2, but you are missing the point.  

Why are you getting this error?  Most likely the Talon 15 is not on the bus at all.  
Use self-test to sanity check your hardware/connections....
https://github.com/CrossTheRoadElec/Phoenix-Documentation/blob/master/README.md#check-the-web-based-configuration
...Use self-test to sanity check the sensor.

HannahGuo, 2018-01-15:
Is it possible to reset Talons wired in CAN to their default state? We would like to reset the Talon control period and all the Talon status frame periods.

Each calibration we've tried has failed, and after looking at last year's Talon SRX User Guide, we discovered calibration was for PWM.

ozrien, 2018-01-15:
Status frame periods default on power cycle. They are not persistent, that's why they are "set" and not "config" functions.

Control frame defaults to 10ms when you construct a Talon object.

The B/C CAL button defaults all configuration parameters (config* functions) as these are persistent.  Also this requires new firmware (3.X) packaged with Phoenix.

AaronPinto, 2018-01-15:
@ozrien would I also get that error if the Mag Encoder's LED is red/switches from yellow to red multiple times or is it only if the Talon stops receiving power? Also, do Talon's contain an auto-resetting internal thermal/current breaker?

ozrien, 2018-01-15:
@HannahGuo What is the exact problem you are seeing with your robot?  Perhaps contacting support@ctr-electronics.com might help directly root cause the problem.

HannahGuo, 2018-01-15:
@ozrien -- @AaronPinto and I are having the same problem.

ozrien, 2018-01-15:
@AaronPinto Mag encoder LED only reflects the mag encoders detection of the magnet, see mag encoder user's guide....
http://www.ctr-electronics.com/srx-magnetic-encoder.html#product_tabs_technical_resources
See section 1.6.

Talons have no need for a resetting thermal breaker.  But your PDP has breakers.  Use the Driver Station Current Log Viewer or listen for the breaker if you are attempting to prove a breaker trip event.

Talons do have current-limiting features than can reduce the motor output to hold a current limit.

ozrien, 2018-01-15:
@HannahGuo If you are receiving "CAN frame not received/too-stale" in the driver station, please use the self-test feature of the web-based config to ensure the Talon is present on the bus, has the correct ID, and is reading proper sensor values.
https://github.com/CrossTheRoadElec/Phoenix-Documentation/blob/master/README.md#check-the-web-based-configuration

HannahGuo, 2018-01-15:
@ozrien Alright, we've fixed the problem. Thank you for your help!

AaronPinto, 2018-01-15:
@ozrien Thank you very much!

ozrien, 2018-01-19:
Defaults Status Frame Periods are available in the updated software reference manual.

[image]

https://github.com/CrossTheRoadElec/Phoenix-Documentation/blob/master/Talon%20SRX%20Victor%20SPX%20-%20Software%20Reference%20Manual.pdf
and is also available....
http://www.ctr-electronics.com/talon-srx.html#product_tabs_technical_resources under Software Manual.

closed 2018-01-19 by ozrien