CrossTheRoadElec/Phoenix5-Documentation #11: Link to new Firmware

team7116, 2018-01-12:
I don't see where the new firmware is located. Is there a link on the web?

In my public documenst I only have the 2.00 version of the Talon SRX. I can't locate the new version.

nix342, 2018-01-12:
The new version is installed with the new Phoenix Framework. The documentation with the location to find it is here:
https://github.com/CrossTheRoadElec/Phoenix-Documentation#update-your-can-devices

JCaporuscio, 2018-01-12:
https://github.com/CrossTheRoadElec/Phoenix-Documentation#update-your-can-devices

Jacob Caporuscio
Software/Hardware Engineer
CTR Electronics

ozrien, 2018-01-13:
Did you run the Phoenix installer?  Talon 2.0 is from two seasons ago.

Tommsy64, 2018-01-15:
Download and run the CTRE Phoenix Framework Installer [here](http://www.ctr-electronics.com/control-system/hro.html#product_tabs_technical_resources). The firmware `.crf` files should be located in the `C:\Users\Public\Documents\Cross The Road Electronics\LifeBoat\HERO Firmware Files` folder.

ozrien, 2018-01-19:
The firmware is on the site, installed on the PC, and doc'd in several places so closing the issue.

closed 2018-01-19 by ozrien