CrossTheRoadElec/Phoenix5-Documentation #12: Version number?

duane701, 2018-01-13:
How can I tell which version of the software is installed? I'm on Mac OS. 

I was hoping to find a version number in a header file somewhere.

How do I know I'm up to date?

JCaporuscio, 2018-01-13:
Unfortunately there's no version number built into the software API.  This has been on our to-do list for a while, but the core library features had to be prioritized.

However, in the non-windows zip file you probably downloaded there's a file called VERSION_NOTES.  While it doesn't have an extension, it can be opened in any standard text editor.
It contains all of the version strings for the associated Phoenix libraries within the zip.

For now, I would recommend placing this file next to the binary installations so you can reference your version numbers if necessary.

JCaporuscio, 2018-01-31:
This is now documented.

https://github.com/CrossTheRoadElec/Phoenix-Documentation/commit/708df6052cb71653078c4e04312bf04c39130701

closed 2018-01-31 by JCaporuscio