CrossTheRoadElec/Phoenix5-Documentation #16: What exactly is the ordinal parameter in the Param API? (And other questions)

Tommsy64, 2018-01-15:
At first glance, the ordinal parameter appears to be the profile slot in which to store and retrieve the parameters? (For instance, eProfileParamSlot_P) Yet, the value appears to be ignored by other parameters such as eMotMag_Accel.

How can I tell when the ordinal is ignored? Or when the `subValue` is ignored in `configSetParameter`?

Since there are only 4 (?) profiles, what is happening when I set the eProfileParamSlot_P parameter, for instance, with an ordinal of 12? Shouldn't an error be thrown? Instead, the ErrorCode returned is `OK`. However, for an ordinal value greater than 15 a `CAN_INVALID_PARAM` ErrorCode is returned.

The following
```
for (int i = 0; i < 50; i++)
    System.out.println("Slot: " + i + " Status: " + talon.config_kP(i, 10 + i, 10).name());
```
outputs:
```
...
Slot: 14 Status: OK 
Slot: 15 Status: OK 
ERROR  -2  CTR: Incorrect argument passed into function/VI.  Talon SRX 3 Config_kP 
ERROR  -2  CTR: Incorrect argument passed into function/VI.  Talon SRX 3 Config_kP 
Slot: 16 Status: CAN_INVALID_PARAM 
Slot: 17 Status: CAN_INVALID_PARAM 
...
```

Reading the eProfileParamSlot_P parameter using `configGetParameter` only increases my confusion. I set the P value and read it like so:
```
for (int i = 0; i < 25; i++)
    System.out.println("Ordinal: " + i + " Status: " + talon.config_kP(i, 10 + i, 10).name());

for (int i = 0; i < 25; i++)
    System.out.println("Ordinal: " + i + " Value: " + talon.configGetParameter(ParamEnum.eProfileParamSlot_P, i, 10));
```

I am able to **set** the first **15** ordinals (or profile slots?).
```
...
 Ordinal: 15 Status: OK 
ERROR  -2  CTR: Incorrect argument passed into function/VI.  Talon SRX 3 Config_kP 
ERROR  -2  CTR: Incorrect argument passed into function/VI.  Talon SRX 3 Config_kP 
 Ordinal: 16 Status: CAN_INVALID_PARAM 
...
```

Next, I can **read** the first **15** ordinals.
```
 Ordinal: 0 Value: 10.0 
 Ordinal: 1 Value: 11.0 
 Ordinal: 2 Value: 12.0 
 Ordinal: 3 Value: 25.0 
 Ordinal: 4 Value: 25.0 
 Ordinal: 5 Value: 25.0 
 Ordinal: 6 Value: 25.0 
 Ordinal: 7 Value: 25.0 
 Ordinal: 8 Value: 25.0 
 Ordinal: 9 Value: 25.0 
 Ordinal: 10 Value: 25.0 
 Ordinal: 11 Value: 25.0 
 Ordinal: 12 Value: 25.0 
 Ordinal: 13 Value: 25.0 
 Ordinal: 14 Value: 25.0 
 Ordinal: 15 Value: 25.0 
ERROR  -2  CTR: Incorrect argument passed into function/VI.  Talon SRX 3 ConfigGetParameter 
ERROR  -2  CTR: Incorrect argument passed into function/VI.  Talon SRX 3 ConfigGetParameter 
 Ordinal: 16 Value: 0.0 
 Ordinal: 17 Value: 0.0 
...
```
But it appears that ordinal values are clamped to a maximum of 3 when reading the parameter??? Yet an error is only caused when the ordinal is greater than 15. Shouldn't such behavior be documented?

Similarly, shouldn't there be an error thrown when invalid profile slots or PID slots are selected using `selectProfileSlot`?

As you can tell, I am greatly confused. Some clarification would be appreciated.

ozrien, 2018-01-15:
- Ordinal represents the slot, which is valid within [0,3].  Slots 0 and 1 are also view-able in the RIO web-based config.
- Slots [4,15] are technically valid as far as the framing is concerned, but unsupported in this season's firmware.  I agree typically we would yield an error code outside of [0,3].  But that is typically a limitation of the firmware, not the API.

Currently if a slot number that is too big is passed into the set, it defaults to 3.  That's why you read 25 on slot 3 after attempting to set a value of 25 on slot15 (which is invalid).

Tommsy64, 2018-01-15:
Ok, thank you. Does that mean `selectProfileSlot(slotIdx, pidIdx)` clamps slotIdx to [0,3] and pidIdx [0,2] without indication? Perhaps adding this to the documentation would be beneficial. Such behavior is not obvious without experimentation.

Likewise, is there a place where I can figure out what the ordinal for each value in ParamEnum does?