CrossTheRoadElec/Phoenix5-Documentation #17: Ability to check limit switch status in code?

wrall, 2018-01-15:
Is there (or, could there be) an ability to check the current limit switch status for the forward and reverse limit switches?  It would be nice to have a function like getForwardLimitSwitchStatusPressed() and getReverseLimitSwitchPressed() to show whether the forward/reverse limit switches are pressed or not.

ozrien, 2018-01-16:
https://github.com/CrossTheRoadElec/Phoenix-Documentation/blob/master/Migration%20Guide.md

Search for "isFwdLimitSwitchClosed"

This is also useful if you just want to add some IO to the Talon (not necessarily for limit switch behavior).  Think of each Talon as an additional analog/digital/quad/pulse-width port so you don't have to hang every sensor off the RIO.

wrall, 2018-01-16:
Ah, I see.  Perhaps that could be mentioned briefly in the main readme?

Thanks!

JCaporuscio, 2018-01-31:
This is now mentioned in the main readme.

https://github.com/CrossTheRoadElec/Phoenix-Documentation/commit/708df6052cb71653078c4e04312bf04c39130701

closed 2018-01-31 by JCaporuscio