CrossTheRoadElec/Phoenix5-Documentation #18: How to use setSensorPhase and setInterval

nix342, 2018-01-16:
Did some experimentation and testing with setSensorPhase and setInterval today and had a hard time making sense of them. With both set to false using velocity control mode, the right and left sides of our drive behave similarly with the exception that the sensor for the left side reads negative when going forward. We wanted to try to resolve this, so we used setSensorPhase(true). With only that single change, we are able to see the sensor readings heading in the correct positive direction, but the behavior of the drive is very strange. The left side behaved as if there was a fairly high ramp rate. With setSensorPhase set back to false, both left and right behave the same...quite responsive. We were at a loss to try to explain why changing the sensor phase had that kind of impact on the drive performance.

Full code is here...it's current state is such that forward is negative for the left side and positive for the right:
https://github.com/frc-88/2018-TestBot/blob/master/src/org/usfirst/frc/team88/robot/subsystems/Drive.java

We ended up giving up after trying many combinations of setSensorPhase and setInverted and just accepted the fact that negative is forward on the left side for now, but we'd like to get this figured out and understand these methods of the new API.

Additional documentation and a procedure for properly configuring and tuning PID (as is found in the old pdf documentation) would be welcome.

Thanks!

ozrien, 2018-01-16:
C++ Example already exists...
https://github.com/CrossTheRoadElec/Phoenix-Examples-Languages/blob/master/C%2B%2B/DifferentialDrive/src/Robot.cpp

Your observation that one side is red (negative sensor) when moving forward sounds correct.  In the WPILIB Differential Drive, the implementation seems to assume that the motors need opposite polarities...
https://github.com/wpilibsuite/allwpilib/blob/1f4822f33278edbbe4d166b7b9baa0bade8c1c4d/wpilibj/src/main/java/edu/wpi/first/wpilibj/drive/DifferentialDrive.java#L264
...which is not guaranteed on all diff drive robots, and forces you to command one side to be negative when moving forward.  Ideally they should have just let the motor controller Inverts do what they are supposed to do.

But as long as the sensor is in phase on both sides, and you track which side has to be negative, it is workable.

The walk-through documentation from last year still generally applies, though we are updating and will release the PDF to be Phoenix compliant.  Tuning procedure is still the same.

ozrien, 2018-01-16:
@nix342 setInterval was supposed to be setInverted right?

nix342, 2018-01-16:
That's right...I've fixed it.

nix342, 2018-01-23:
I think I just discovered what our issue was. Just found this is the documentation:
"Inverting Followers
When using motor controllers in a master/follower configuration, all motor controllers need to have their direction set independently. A follower controller will not automatically change direction if you only invert the master."

I don't think we were doing this...and that would do a lot to explain the behavior we were seeing. We will try to do this correctly tonight and see if that was our problem.

nix342, 2018-02-22:
Our issue was, in fact, due to not setting invert on the followers thinking they would mimic the master. I now see this is a desirable feature as sometimes followers will be inverted from masters. This issue can be closed.

closed 2018-02-22 by nix342