CrossTheRoadElec/Phoenix5-Documentation #19: Replacement for enableZeroSensorPositionOnForwardLimit()?

PChild, 2018-01-17:


JCaporuscio, 2018-01-17:
There's no API function right now, but you can use configSetParameter.

Use the eClearPositionOnLimitF parameter.  Pass in a value of 0 to disable the feature and a value of 1 to enable.  subValue and ordinal can be zero.

So the following would enable zeroing the sensor position on forward limit (with a timeout of 10 ms for the set):

talon.configSetParameter(ParamEnum.eClearPositionOnLimitF, 1, 0, 0, 10);

The same method can be used for clearing position on reverse limit or index.

PChild, 2018-01-17:
Thanks for the help. Should that info go in the Migration Guide?

ozrien, 2018-01-19:
Feature explanation and examples are available in the update software reference manual.

[image]

[image]

http://www.ctr-electronics.com/talon-srx.html#product_tabs_technical_resources or https://github.com/CrossTheRoadElec/Phoenix-Documentation

closed 2018-01-19 by PChild