CrossTheRoadElec/Phoenix5-Documentation #22: Java Source Attachment

bradfrc1124, 2018-01-21:
The section with screen shots of how to configure the Java Source Attachment does not match the User Interface. Could this be updated to show the new way to do it? I can't figure it out.

ozrien, 2018-01-21:
Looks correct to me (Windows Eclipse Oxygen and Windows Eclipse Mars).  
@bradfrc1124  You need to post screenshots of where there is a discrepancy.

bradfrc1124, 2018-01-21:
Thank you for quick reply. 

[image]

bradfrc1124, 2018-01-21:
And for comparison, this is from the doco:

[image]

bradfrc1124, 2018-01-21:
My software version:
Eclipse IDE for Java Developers
Version: Oxygen.2 Release (4.7.2)
Build id: 20171218-0600

ozrien, 2018-01-21:
My eclipse has the same version 20171218-0600, 4.7.2, but it is Eclipse IDE for C/C++ with Java added via plugins.  Maybe that's the difference.

Regardless, your interface looks even simpler.  Just path it to CTRE_Phoenix-sources.jar, similarly to step 4 [under here](https://github.com/CrossTheRoadElec/Phoenix-Documentation#add-javadoc-if-using-java).

If that's too difficult because of the workspace relative pathing, select **external location** and give the same path to CTRE_Phoenix-source.jar that you would have provided in the documented interface.

As long as eclipse can find CTRE_Phoenix-sources.jar, the intellisense features should work.

bradfrc1124, 2018-01-21:
Thanks! That worked (kind of). I've added the TalonSRX element into the Java code and I was able to import the motor control package. When I hover over TalonSRX, I still get this:

_Note: This element has no attached Javadoc and the Javadoc could not be found in the attached source._

I am guess that just means there is no doc for that specific element.

closed 2018-01-21 by bradfrc1124