CrossTheRoadElec/Phoenix5-Documentation #26: CTRE CAN Receive Timeout

martygubow, 2018-01-28:
ERROR  1  CTRE CAN Receive Timeout  java.lang.Thread.run(Thread.java:748)

I cannot seem to get rid of this message on Driverstation.   I have removed all references to the Talons in my code.  Removed every import and commented out all Talon code.  The message still shows up?

ozrien, 2018-01-28:
[1] Make sure your code changes are making to the robot controller.  Add a System.out.println call or something to convince yourself your changes are taking place.
[2]  That error messages doesn't quite match the typical Phoenix [formatted error ](https://github.com/CrossTheRoadElec/Phoenix-Documentation#driver-station-says-the-firmware-is-too-old) which includes the device and ID.  That could be PDP or PCM related (which is still part of wpilib).

martygubow, 2018-01-29:
Omar,

Thank you for getting back to me.   My changes are definitely being
applied.  I get the error once a second.  I actually  stated a new project
and tested it at each step.  As soon as the Phoenix initialization is
complete the error start.*   I update all of the FRC software.  I am
running the latest Phoenix, Driverstation, WPLIb etc.

* Note I did not get the error before I put in the code to communicate with
the Talon.   Once I did the error started and don;t stop

Thank you for your help

Marty

martygubow, 2018-01-29:
FYI.. just found a thread on Chief Delphi

The error is being caused by Telemetry

I added the command below in Robot Init and the problem stopped

LiveWindow.disableAllTelemetry();

JCaporuscio, 2018-01-31:
Closing issue.

closed 2018-01-31 by JCaporuscio