CrossTheRoadElec/Phoenix5-Documentation #28: Question about Motion Magic Setup

ShreyaRavi, 2018-02-01:
I was wondering if there's something wrong with the way I am setting up my program to use Motion Magic.

In `robotInit()`, I am calling the following methods:
`setInverted()` -- set to `false` (after testing)
`setSensorPhase()` -- set to `false` (after testing)
`configSelectedFeedbackSensor()` -- set to `FeedbackDevice.QuadEncoder`
`configNominalOutputForward()` -- set to `0`
`configNominalOutputReverse()` -- set to `0`
`configPeakOutputForward()` -- set to `1`
`configPeakOutputReverse()` -- set to `-1`
`setSelectedSensorPosition(0)`

and I set the PIDF values (still in `robotInit()`) using:
`config_kP()`
`config_kI()`
`config_kD()`
`config_kF()`
(all of the PID values are set to 0 initially and the Feed Forward Gain is calculated as discussed further below)

and I set the cruise velocity and acceleration (also in `robotInit()`) using:
`configMotionCruiseVelocity()`
`configMotionAcceleration()`

Then, in `teleopUpdate()`, I call the `set()` method as follows:
1. Calculate the number of native units to move forward --> `number_of_native_units`
2. Call `set(ControlMode.MotionMagic, number_of_native_units)`

I calculated the Feed Forward Gain as follows (using the Software Reference Manual's steps):
1. The maximum speed for the wheels (which are a geared output moving a ~120 lb load) was measured through testing to be 14.2 ft/s.
2. This was converted to revolutions per 100 ms on the motor output (which is what the encoder is attached to). With a gear ratio of 8.45:1 and a wheel diameter of 6 in, this was (14.2 ft/s * 12 in/ft * 1 revolution/6π in *

JCaporuscio, 2018-02-01:
This sounds like a support request.

Please send an e-mail to support@ctr-electronics.com.
In your e-mail, please describe the behavior you are seeing vs. what you are expecting.

closed 2018-02-01 by ShreyaRavi

team6637, 2018-02-13:
Same Issue here! Any resolution?