CrossTheRoadElec/Phoenix5-Documentation #35: Can't find instructions on how to set a velocity-only trajectory point for Motion Profile

nestor-sg, 2018-03-06:
In the older documentation you were able to specify a Motion Profile trajectory point as velocity-only with a boolean flag.

In the newer Phoenix documentation, there is only one function to set a trajectory point but the velocity-only flag is no longer available.

Can your team add instructions to the documentation as how one deals with a velocity-only trajectory point? 

I'm under a long guess here to say that you set your F gain, leave all other PID parameters as zero (0) and that would make the function to ignore whatever you input in position and heading inputs.

Thanks!
Nestor (Team 4343)

ozrien, 2018-03-06:
Correct, zero your kP,Ki,kD and leave your kF nonzero.  That is the equivalent as last-season's vel-only flag.
Additionally, this season you have two more slots for a total of four [0,3].  So you likely can leverage this for vel-only equivalent use.

nestor-sg, 2018-03-07:
Thanks Omar, will do... do you want to keep this issue open so this can be added to the documentation later? Or do you wish to close it?

closed 2020-01-09 by JCaporuscio