CrossTheRoadElec/Phoenix5-Documentation #36: Add a section for API performance expectations

ozrien, 2018-03-11:
Our API performance expectations are....
- ~0.3ms per call for any get* routines.
- ~0.3ms per call for any set* or enable* routines where the input has changed since previous call.
- ~0.0ms per call for any set* or enable* routines where the inputs have not changed since previous call
- ~ 4ms for any successful ConfigXXXX routines if non zero timeoutMs is passed.  These should be done on robot boot.
- ~ Xms for any timed out ConfigXXXX routines if X timeoutMs is passed.  These should be done on robot boot. 
- ~ 0.3ms for any ConfigXXXX if zero timeoutMs is passed.  These should be done in the robot loop, if at all (generally not necessary).  Success is not determined since there is no wait-for-response checking.
- ~ 4ms for any successful ConfigGetXXXX.  These are generally not necessary.

The ~0.3 ms time is a limitation of the roboRIO.  Many of the functions in WPILIB have this limitation.  For example, accessing joystick information requires ~0.3ms per call.

ddyer8, 2018-03-12:
Is this true for JNI calls too? We're running java and in a test I was running the other day I was seeing more like 1-1.5ms for a get* call and sometimes as long as maybe 4-5ms.

I was looking into this because I'm writing telemetry files from my robot.  I was noticing at 50Hz (20ms) I couldn't capture more than about 10 parameters from Talons.

ozrien, 2018-03-15:
Which calls?  That sounds like a ConfigGetXXXX, which would be about 4ms.

ozrien, 2018-04-03:
Section added...
https://github.com/CrossTheRoadElec/Phoenix-Documentation#performance-expectations-for-api-calls
...so closing

closed 2018-04-03 by ozrien

ddyer8, 2018-04-04:
I meant to follow up from my above comment. I was calling get* and not config* methods. However, when I attempted to reproduce in a standalone project, I got the performance you indicated: <= 0.3ms. I believe we are experiencing longer delays in my project because of high CPU usage and/or thread contention.