CrossTheRoadElec/Phoenix5-Documentation #42: Complete C++ documentation

kebrx, 2018-11-24:
I'm confused about if your C++ API has all functionality implemented or if the documentation is incomplete.  

Please complete the documentation or the API.

ozrien, 2018-11-24:
There will be  various documentation updates throughout December for the features we added in the off season- but its not clear to me what C++ API you are confused about.  Please point these out in the issue tracker, otherwise the tracker is not useful.

In the meantime, if you have API questions you can ...
- ask here 
- contact us: support@ctr-electronics.com
- review our many GitHub examples
- The [readme ](https://github.com/CrossTheRoadElec/Phoenix-Documentation/blob/master/README.md)has a general walk-through, and top top banner has more-detailed oriented documentation, look for "2018 Phoenix Documentation | Migration Guide | API Docs Java C++" links.
- The [software ref manual ](https://github.com/CrossTheRoadElec/Phoenix-Documentation/raw/master/Talon%20SRX%20Victor%20SPX%20-%20Software%20Reference%20Manual.pdf) has the entire API explained.

JCaporuscio, 2020-01-09:
Closing as this issue is out-of-date.
If you feel there are documentation points that need clarification addition, please create another issue tracker or email support@ctr-electronics.com

closed 2020-01-09 by JCaporuscio