CrossTheRoadElec/Phoenix5-Documentation #61: Documentation of Library Installation seems to be wrong

Lytigas, 2020-01-20:
https://phoenix-documentation.readthedocs.io/en/latest/ch05a_CppJava.html

> Alternatively you can use “Install new libraries (online)” option with http://devsite.ctr-electronics.com/maven/release/com/ctre/phoenix/Phoenix-latest.json. However this is not recommended as this requires a live Internet connection to use your FRC project.

This doesn't seem to be correct. Per https://docs.wpilib.org/en/latest/docs/software/wpilib-overview/3rd-party-libraries.html the distinction is where the VSCode extension looks: either for a pre-installed library already on the users computer, or from a URL for a JSON file. My undestanding is that install online or offline only affects the first build, after which the cache is used. Moreover, unless you start your project on a computer with the Phoenix framework installed (ie by the executable installer) an online install is the only way to get the library. For instance, my development machine is completely separate from the driver station, which hosts all the NI and Phoenix configuration software.

JCaporuscio, 2020-01-20:
>My undestanding is that install online or offline only affects the first build, after which the cache is used.

The cache is controlled by gradle, not by CTRE or by WPILib.  In the past WPILib has had to put in workarounds and request modifications to gradle itself to attempt to prevent the cache from being automatically cleared after a period of time.  In theory the cache prevents the need to re-download dependencies, but there's no guarantee of this behavior.

Each year there are new issues found when downloading maven dependencies from a remote instead of a local install, including loss of internet during initial download (requires manually clearing gradle cache to fix) or having *any* online maven repo unavailable.

Because of this we don't consider the gradle cache a reliable method of offline use.
We recommend having *all* required maven components (WPILib, CTRE, and other vendors) installed locally.

> Moreover, unless you start your project on a computer with the Phoenix framework installed (ie by the executable installer) an online install is the only way to get the library. For instance, my [...]

Phoenix should be installed on any machine being used for development, not just the driver station.  The NI components are not a requirement for installing Phoenix.

For any Windows machine the .exe can be used, and for any linux/mac installation the non-Windows zip can be used to place the components on the local machine.
For example, both of these are available with the latest release (and have equivalent links on our website): https://github.com/CrossTheRoadElec/Phoenix-Releases/releases/tag/v5.17.6

Lytigas, 2020-01-23:
Thanks for clarifying that. What you say makes perfect sense. It seems that the WPILib docs might be wrong:

> The libraries themselves are placed in the Maven cache at C:\Users\Public\wpilib\YYYY\maven. Vendors can place a local copy here with an offline installer (recommended) or require users to be online for an initial build to fetch the library from a remote Maven location.

This implies that the libraries for an online install are then places there, but I can't actually find any Phoenix stuff in there even after building my online-installed projects.

JCaporuscio, 2020-01-23:
What you're observing is correct.

The online build only stores the artifacts in the local gradle cache, it does not permanently download them into the wpilib install location.

In *most* cases using the online version is OK as the gradle cache is *usually* fine, but there are too many edge cases/odd behavior for us to recommend the online version for Phoenix users.

closed 2020-01-24 by Lytigas