CrossTheRoadElec/Phoenix5-Documentation #63: Feature request: Documentation: Full TalonFX Current Limit API guide

samneff72, 2020-01-27:
# Feature Requests

## Feature request in as much detail as possible.

Update "New API in 2020" section at https://phoenix-documentation.readthedocs.io/en/latest/ch13_MC.html#new-api-in-2020 to include complete details (descriptions & typical values) associated with common settings available in the TalonFX Current Limiting API, or at least direct links to code examples that work for supply-limiting in 2020. 

Stretch goal: complete details for all settings for TalonFX Current Limiting on both supply & stator sides

## Why it is beneficial.

The existing (1/27/2020) public-facing current limiting documentation does not have a complete list of settings & common values required to make the feature work in 2020 on the TalonFX API. 

https://phoenix-documentation.readthedocs.io/en/latest/ch13_MC.html#new-api-in-2020

The existing public-facing current limiting code examples are for the 2019 API

https://github.com/CrossTheRoadElec/Phoenix-Examples-Languages/tree/master/Java/Current%20Limit/src/main/java/frc/robot

https://github.com/CrossTheRoadElec/Phoenix-Examples-Languages/tree/master/Java/CurrentClosedLoop/src/main/java/frc/robot

The ability to limit current on the TalonFX (similar to the featureset available on the TalonSRX) was a significant part of our team's buying decision on the Falcon500. If the feature is available but is not documented for us to activate, it might as well not be available. 

## A series of steps to test the feature after it is implemented

Documentation update

samneff72, 2020-01-27:
Relevant Chief thread https://www.chiefdelphi.com/t/current-limiting-talonfx-values/374780/11

JCaporuscio, 2020-01-27:
The Supply Current Limit for Talon FX is identical to the traditional current limiting in the Talon SRX.

The only difference is that the parameters are bundled into a single configuration object so only a single API call is needed.

For Java:
API Call: http://www.ctr-electronics.com/downloads/api/java/html/classcom_1_1ctre_1_1phoenix_1_1motorcontrol_1_1can_1_1_talon_f_x.html#a302c7824cf2728ee03f1b739a14e2a35
Config Object:http://www.ctr-electronics.com/downloads/api/java/html/classcom_1_1ctre_1_1phoenix_1_1motorcontrol_1_1_supply_current_limit_configuration.html

These API docs are also available in the intellisense (or javadoc jar from the installer/maven) in vscode.

These functions are listed in the doc, but I can add a code snippet of the API call.  
Beyond this, is there any other point of confusion with the new API?

samneff72, 2020-01-27:
That's rad, definitely better than last year. Good change. Thanks for the
direct link into the API call & config object for java!

I agree we should add a code snippet or two in the doc, I think that'll
make it much easier for students to understand and work with instead of
forcing them to dig into the API.... and giving them a baseline value to
start with would be really helpful.

//One common supply-side current limiting case is to enable the feature to
prevent a TalonFX controller from pulling more than 45A from the PDP,
starting when any current flows (0A threshold) immediately (0ms after
threshold)
boolean enable = true; // enable feature
double currentLimit = 45; // limit current to this value, in amps
double triggerThresholdCurrent = 0; // threshold current to begin limiting
after, in amps
double triggerThresholdTime = 0; // time in milliseconds after
threshold current is observed to begin limiting
talonFX.configSupplyCurrentLimit(enable, currentLimit,
triggerThresholdCurrent, triggerThresholdTime)

The really accessible code examples for the TalonSRX current limiting in
the SRX User Guide were a fantastic asset when we started using them a few
years ago, looking for something similar here.

Best Regards,
Sam Neff

bhall-ctre, 2026-09-07:
This has been addressed in Phoenix 6: https://v6.docs.ctr-electronics.com/en/stable/docs/hardware-reference/talonfx/improving-performance-with-current-limits.html

closed 2026-09-07 by bhall-ctre