CrossTheRoadElec/Phoenix5-Documentation #69: Can frame not received/too error with a CTR Magencoder and a velocity CLosed Loop

maxeitelwein, 2020-02-29:
private final WPI_TalonSRX flywheelMotor1 = new WPI_TalonSRX(Constants.Motors.FLYWHEEL_1);
private final WPI_TalonSRX flywheelMotor2 = new WPI_TalonSRX(Constants.Motors.FLYWHEEL_2);
flywheelMotor2.configFactoryDefault();
flywheelMotor1.configFactoryDefault();
		
flywheelMotor1.setInverted(true);
flywheelMotor2.setInverted(false);
		
flywheelMotor1.setNeutralMode(NeutralMode.Coast);
flywheelMotor2.setNeutralMode(NeutralMode.Coast);

/* Config sensor used for Primary PID [Velocity]*/
		flywheelMotor1.configSelectedFeedbackSensor(FeedbackDevice.CTRE_MagEncoder_Relative,
		0,
		Constants.Autonomous.TIMEOUT);
		
flywheelMotor1.setSensorPhase(false);
		
flywheelMotor1.configPeakOutputForward(1, Constants.Autonomous.TIMEOUT);
flywheelMotor1.configPeakOutputReverse(-1, Constants.Autonomous.TIMEOUT);
flywheelMotor2.configPeakOutputForward(1, Constants.Autonomous.TIMEOUT);
flywheelMotor2.configPeakOutputReverse(-1, Constants.Autonomous.TIMEOUT);
		
flywheelMotor1.config_kF(0, kF, Constants.Autonomous.TIMEOUT);
flywheelMotor1.config_kP(0, kP, Constants.Autonomous.TIMEOUT);
flywheelMotor1.config_kI(0, kI, Constants.Autonomous.TIMEOUT);
flywheelMotor1.config_kD(0, kD, Constants.Autonomous.TIMEOUT);
flywheelMotor1.configClosedloopRamp(1, Constants.Autonomous.TIMEOUT);

public void setFlyWheel(){
    flywheelMotor1.set(ControlMode.Velocity, rpmout);
    flywheelMotor2.follow(flywheelMotor1);

We are running two 775 pros at around 6000RPM (yeah it is fast but that should not be a problem)
The error accures after about 3 seconds of spinning

maxeitelwein, 2020-02-29:
This is what is confusing to me: We are drawing 50amps(!) and have a bus voltage of 8.5V when we are up to speed.

maxeitelwein, 2020-02-29:
Fixed it. The motors tried to accelerate into oblivion and then a braker popped.

closed 2020-02-29 by maxeitelwein

ozrien, 2020-02-29:
Always check the self-test first when troubleshooting a CTRE Motor Controller.  There is a fault for detecting this exact root-cause ("reset during enable").