CrossTheRoadElec/Phoenix5-Documentation #70: Typo in picture

Starlight220, 2020-04-30:
In [this](https://phoenix-documentation.readthedocs.io/en/latest/ch05a_CppJava.html) page, the picture showing a Java compilation error (as an example of the library not loaded), the error is because of the compiler doesn't find a `TalonSSRX` (emphasis on the double `S`), and not because of the library not being loaded properly. 
The C++ example is clear on the error is because of library problems, however, the Java problem has nothing to do with the library. 

Is this intentional (the java error as an example of fixing a compilation error), or is this an oversight/typo (the error should be because of lib errors)?

JCaporuscio, 2020-04-30:
This is intentional.  The java picture you're referencing is just an example of a generic compiler error for the purposes of the walkthrough, specifically to illustrate navigating to syntax errors using the terminal output.

Is there something in the documentation that made you think it was supposed to illustrate a library not loaded issue?

Starlight220, 2020-04-30:
The C++ picture is an example of a library error. I thought about C++/Java parallelism. It might have simply been my misunderstanding, with no need of changing anything.

closed 2020-04-30 by Starlight220