CrossTheRoadElec/Phoenix5-Documentation #94: Add link to Java/C++ javadocs to homepage

Daltz333, 2022-01-24:
Finding the javadocs was quite difficult. It would be useful to add these to the homepage or sidebar.

JCaporuscio, 2022-01-25:
Agreed.
There's some larger restructure that still needs to happen (see https://github.com/CrossTheRoadElec/Phoenix-Documentation/issues/78) but we should be able to split the API doc links out to be more obvious without waiting on that.

ozrien, 2022-01-26:
Agreed.  
It's also listed here at the bottom:  https://store.ctr-electronics.com/software/

Daltz333, 2022-01-26:
Yeah, I found that after someone pointed it out to me.

JCaporuscio, 2022-05-10:
Just realized I never closed this - I added the C++/Java API docs links directly to the ToC tree so they should be easier to find.
https://github.com/CrossTheRoadElec/Phoenix-Documentation/commit/8ff6087ad3d43c6b0be74bac650a8cff93a12111

closed 2022-05-10 by JCaporuscio