CrossTheRoadElec/Phoenix5-Documentation #97: LabVIEW Snippet Doesn't Work

JonathanLindsey, 2022-02-21:
## [1] Describe your setup.
1. Tested with FireFox and Chome
2. LabVIEW 2020 (for FRC 2022)
2. FRC Game Tools 2022 f1
3. Phoenix software

## [2] List firmware versions for ALL CTRE devices and your Phoenix Installation.
N/A

## [3] What behavior are you expecting and why?
The LabVIEW snippet from https://docs.ctre-phoenix.com/en/stable/ch11_BringUpPigeon.html#pigeon-api should drag/drop into a block diagram and turn into code.

The `Tip` just above the snippet says:
```
Image below can be dragged/dropped into LabVIEW editor.
```

**Note: I have not looked to see if there are other LabVIEW snippets in the documentation and if they work or not.**

## [4] What behavior are you observing (and how is it different from your expected behavior above)?

Depending on the method used, either a link or image is shown in the LabVIEW block diagram but not the actual code.

## [5] List the procedure to reproduce the issue.

I tested each method in FireFox and Chrome

Method 1:

1. Dragging directly from the browser to a block diagram
2. A LabVIEW comment with a link is added to the block diagram

Method 2:

1. Opening the image in a new tab and dragging from there into a block diagram
2. A LabVIEW comment with a link is added to the block diagram

Method 3:

1. Right click image
2. Save image as...
3. Drag the image from File Explorer to a block diagram
4. The image is shown in LabVIEW (but it is not code)

JonathanLindsey, 2022-02-21:
There was a similar problem on the ChiefDelphi site itself. It was solved by changing a Discourse setting to not strip image metadata.

https://www.chiefdelphi.com/t/labview-snippets-not-working-on-new-cd/344342/3?u=jonathan_l

https://meta.discourse.org/t/site-setting-for-strip-image-metadata/69276

JCaporuscio, 2022-02-21:
We'll take a look and try to reproduce this - it could be related to RTD or to mismatched LabVIEW versions if that's the case.