CrossTheRoadElec/Phoenix5-Documentation #106: return type of `is{Rev,Fwd}LimitSwitchClosed` is inconsistent between TalonFX and TalonSRX

binex-dsk, 2022-05-11:
This function returns an int with the TalonFX, and a boolean with the TalonSRX. These should both be booleans, why is it like this?

JCaporuscio, 2022-05-11:
The `isLimitSwitchClosed` functions return an int in all cases, both for TalonSRX and TalonFX.

They're inherited from a BaseTalon parent class:
https://api.ctr-electronics.com/phoenix/release/java/com/ctre/phoenix/motorcontrol/can/BaseTalon.html#isFwdLimitSwitchClosed()
https://api.ctr-electronics.com/phoenix/release/java/com/ctre/phoenix/motorcontrol/can/BaseTalon.html#isRevLimitSwitchClosed()

Where are you seeing that the SRX version returns a boolean?

binex-dsk, 2022-05-12:
Yeah, I was talking about the `getSensorCollection().is{Rev,Fwd}LimitSwitchClosed()`, didn't realize that this existed.

Thanks for the help.

By the way, "if" is misspelled there.

auscompgeek, 2022-05-13:
> By the way, "if" is misspelled there.

Not a misspelling: https://en.wikipedia.org/wiki/If_and_only_if

binex-dsk, 2022-05-14:
> > By the way, "if" is misspelled there.
> 
> Not a misspelling: https://en.wikipedia.org/wiki/If_and_only_if

Interesting, didn't know that was a thing.